akopachov/flipper-zero_authenticator #1: Can't run the app due to API version mismatch

AngryJKirk, 2022-10-05:
When I run the app I get "Preload failed. API version mismatch"

My firmware version is 0.68.1, stock

AxorPL, 2022-10-07:
Building against the official firmware fails for me, it worked before the repo migration. This is what I get: [totp-official-error.txt](https://github.com/akopachov/flipper-zero_authenticator/files/9731243/totp-official-error.txt)

br0ziliy, 2022-10-07:
You should switch to tag 0.68.1 (`git checkout 0.68.1`) - then the build will succeed. Building against dev branch requires application code update.

AxorPL, 2022-10-07:
My bad, works now, many thanks.

akopachov, 2022-10-07:
Thank you everybody for all these useful info. What I'll try to implement next is two builds, one for official and one for unleashed firmware and do a releases for both of them. Should be not that complicated thing to do I guess.

akopachov, 2022-10-10:
Should be fixed in [latest release](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v0.0.5).
Will be publishing 2 fap files for each release starting from today, one for Unleashed and one for Official firmware.

closed 2022-10-10 by akopachov