akopachov/flipper-zero_authenticator #2: Incorrect OTP generated

vinogradovnet, 2022-10-07:
Can't generate correct OTP for my secret (16 letters)

Time checked - correct

All three SHA1, SHA256, SHA512 tried

All times OTP not match with other generators

How do I fix it?

vinogradovnet, 2022-10-07:
I changed "Timezone:" in totp.conf and OTP generated correctly.

Is it possible to set this value from app?

akopachov, 2022-10-10:
Hey, glad you found solution. I do agree that Flipper Authenticator needs some UI to configure timezone. Also it definitely makes sense to add info to README. Good thing is that I plan to do both :) First of all - will update README, then UI.

akopachov, 2022-10-13:
* README updated to explain better each key in conf file
* Timezone configuration UI was added in [release v0.0.6](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v0.0.6)

closed 2022-10-13 by akopachov