akopachov/flipper-zero_authenticator #7: Change PIN; backup

Gooman-rus, 2022-10-20:
Is it possible to change the PIN?
Is it possible to backup everything via copying config file?

akopachov, 2022-10-20:
> Is it possible to change the PIN?
As of right now it is not possible to change PIN.

> Is it possible to backup everything via copying config file?
It is possible to backup everything via copying config file, but keep in mind that backup file is highly tied to original Flipper where it was created. So if you loose your original device or forget PIN - it will not be possible to use backed up config file on another Flipper.

I'd suggest slightly better way to backup - use your favorite password manager to store plain token secrets. That case if you loose your Flipper, forget PIN or just decide to change it - you would be able to seed your new config file with plain token secrets.

akopachov, 2022-10-20:
Will add this info to README

Gooman-rus, 2022-10-20:
How I can change the PIN?

akopachov, 2022-10-21:
Added [FAQ.md](https://github.com/akopachov/flipper-zero_authenticator/blob/master/FAQ.md) which includes answers to the asked questions.

closed 2022-10-21 by akopachov

Pavgran, 2022-10-22:
Is it really impossible to change pin?
As far as I understand, IV sent to FURI is constructed by XORing base IV with PIN-based value. So, broadly, IV = BIV ^ PIN. Now, if we want to change pin, it's sufficient to change BIV to old_BIV ^ old_PIN ^ newPIN. Now, new_BIV ^ new_PIN = old_BIV ^ old_PIN