akopachov/flipper-zero_authenticator #24: Type code in BadUSB mode enters only last 4 digits and Enter

UndeRus, 2022-11-10:
When I press down button - authenticator enters only last 4 digits and presses Enter key
In some websites you don't need to push enter to enter totp code

akopachov, 2022-11-10:
Thanks for testing. What OS you are running? Agree on "enter" key, probably makes sense to do not type it.

UndeRus, 2022-11-10:
Bug was found on Windows 10

akopachov, 2022-11-10:
Hmmm, is it constantly reproducible?
Asking this because I tried this functionality lot of times on two different laptops running Windows 10, and all the time it was working just as designed. Probably worth it to add 1 sec. delay before simulating key press.

UndeRus, 2022-11-10:
I tried different usb ports:

- on USB 2.0 it always enters 4 digits
- on USB 3.0 it enters 4 digits right after plug in, then on different cables it enters 5 or 6 digits

akopachov, 2022-11-10:
Let's see if [this](https://file.io/bkjw0HSu9pSb) version (with 0.5 sec. delay) acts any better for you.

UndeRus, 2022-11-10:
Sorry, I'm using official firmware 0.70.1, this fap doesn't run

akopachov, 2022-11-10:
Np, [here is](https://file.io/jUrRyyBoZgTe) the one for official firmware.

UndeRus, 2022-11-10:
Thanks, tried this version, both usb 2.0 and usb 3.0 and different cables works stable, app enters 6 symbols any time.

akopachov, 2022-11-10:
Fixed in [v1.2.1](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.2.1)

closed 2022-11-10 by akopachov

mfulz, 2023-11-19:
I've this issue now with the actual version (master RM FW).
Tried different usb ports and cables (usb 2 and 3) got max 5 chars instead of 6

commit e1afd244 (2022-11-10): Fixed #24 (#25)

* Fixed #24
* Small improvement to the USB mode restore code