akopachov/flipper-zero_authenticator #42: Base32 encoded token secret support

ilyabreev, 2022-11-25:
Hi there!

Is there any way to use base32 encoded secret got from `otpauth://totp/smth?secret=xxxxxxx&issuer=zzz` URL
Tried to paste it in config. App runs smoothly but generated codes are wrong.

akopachov, 2022-11-25:
Hey, 
what you need to take out of such a URIs is whatever is going after `secret=` and till `&` or end of string.

So in your example `otpauth://totp/smth?secret=xxxxxxx&issuer=zzz` what you need to provide to Flipper Authenticator is `xxxxxxx` only.

ilyabreev, 2022-11-25:
Thanks for your reply.
Yeah that's exactly what i've done. Also I checked that flipper clock is synced through BT with flipper app.
Trying to login into my Github profile for example. I'm using SHA-1.
Still doesn't work.

I wonder if codes generated with flipper must exactly match with Google Authenticator codes?

akopachov, 2022-11-25:
Have you set correct timezone?

ilyabreev, 2022-11-25:
Yes. It's MSK for my case (+3h).

ilyabreev, 2022-11-25:
Forgot to mention that flipper-zero authenticator refreshes codes slightly later than Google Authenticator app (with 0.5-1.5 seconds delay). May be this an issue?