akopachov/flipper-zero_authenticator #55: Add option to change app pin(Feature request)

crustulumtheoneandonly, 2023-01-02:
Would it be possible to add the option to change the pin in the app settings?
I missed the setup and and dont want to reinstall whole firmware.. (Or can I just delete the settings file?)

Great work!

akopachov, 2023-01-03:
Hey

Once PIN set there is no option to reset\change it without knowing existing PIN and not loosing all the tokens added.

If you know you PIN and want to change it or delete it at all - there is CLI commands: `totp pin set` and `totp pin remove`. This is covered here in [FAQ](https://github.com/akopachov/flipper-zero_authenticator/blob/master/FAQ.md#how-to-changerecover-pin)

If you don't know your existing PIN and you just want to completely reset the app (keep in mind that all tokens you added will gone) - you don't need to reinstall firmware (this will not help actually) instead you may just delete `/ext/authenticator/totp.conf` file. Once you launch Authenticator app next time it will detect that there is no conf file and will start setup from scratch.

Initially I thought about implementing UI to change\remove PIN, but then realized that it is not that frequent usecase to overcomplicate UI and finally I make it as an CLI command for whoever once in a blue moon decide to change\remove PIN.

Thank you for bringing two good points:
1. To explain better in documentation what is going to happen if conf file removed.
2. To add new CLI command `totp reset` to completely reset app to its defaults.

crustulumtheoneandonly, 2023-01-03:
Great. Thanks for this awesome answer!

closed 2023-01-03 by crustulumtheoneandonly