akopachov/flipper-zero_authenticator #64: Support to HMAC-SHA256

auino, 2023-01-17:
I'm trying to use the Flipper Zero in conjunction with [this method](https://github.com/andry08/ArubaOTP-seed-extractor), but provided codes are not correct. I assume it is because `HMAC-SHA256` is not implemented in the app.

akopachov, 2023-01-17:
HMAC-SHA256 is implemented in this app. 

When you are adding new token it is possible to specify which token hashing algorithm should be used. There are 3 possible options `SHA1` (HMAC-SHA1), `SHA256` (HMAC-SHA256) and `SHA512` (HMAC-SHA512).

I'm using [this](https://www.verifyr.com/en/otp/check#totp) website to test all the implemented algorithms.

If you are sure HMAC-SHA256 algo should be used and the codes you see are not correct highly-likely it is because of:
* [Clock on Flipper is not precise](https://github.com/akopachov/flipper-zero_authenticator/blob/master/FAQ.md#clock-is-not-precise)
* [Timezone is not correct](https://github.com/akopachov/flipper-zero_authenticator/blob/master/FAQ.md#timezone-is-not-correct)
* [Token secret is not correct](https://github.com/akopachov/flipper-zero_authenticator/blob/master/FAQ.md#token-secret-is-not-correct)
* Wrong algorithm picked at the step of adding new token. Use CLI command `totp ls` to make sure  that right algo is set for your token.

auino, 2023-01-17:
Thank you. I believe the wrong code is related to the duration of the code, which in my case has to be customizable.

akopachov, 2023-01-17:
Well, if you can prove that custom duration is needed and will solve the problem, then it become a trigger for me. As I was asked multiple time why token duration is statically set to 30 sec, and all the time I answered - "give me any example where 30sec does not work" :) So, if you have some time to prove it and confirm - please let me know and I'll create a new issue to add this custom duration per token

auino, 2023-01-17:
For the mentioned implementation (see link in the first message), the TOTP URI has the following form:

    otpauth://totp/Aruba:userID?secret=[omitted]&issuer=Aruba&algorithm=SHA256&digits=8&period=60

This is working as expected on apps like 1Password, as the `period` field is automatically interpreted. Also, by using the test link provided in the first answer in this thread, I noticed that the provided code depends on the `period`.

akopachov, 2023-01-17:
Thank you for providing this prove and confirming that it definitely should be customizable. It is pretty rare case, but still should be covered.

I'll create a new issue to track this feature request.

closed 2023-01-17 by akopachov

akopachov, 2023-01-18:
@auino this should be implemented in [v1.6.0](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.6.0). Would appreciate if you give it a try

auino, 2023-01-18:
Thanks. It's not working to me and the token is updated every 30 seconds, even by setting `TokenDuration: 60`.

akopachov, 2023-01-18:
Did you put `TokenDuration: 60` manually to conf file?

auino, 2023-01-18:
Yes.

auino, 2023-01-18:
Ok, it is working now, even if the generated code is not correct.

akopachov, 2023-01-18:
There is some progress!
Let's see if one of these would help:
* [Clock on Flipper is not precise](https://github.com/akopachov/flipper-zero_authenticator/blob/master/FAQ.md#clock-is-not-precise)
* [Timezone is not correct](https://github.com/akopachov/flipper-zero_authenticator/blob/master/FAQ.md#timezone-is-not-correct)
* [Token secret is not correct](https://github.com/akopachov/flipper-zero_authenticator/blob/master/FAQ.md#token-secret-is-not-correct)

* Wrong code length (it is possible that your provider requires 8 digit codes, which are also supported by this app)

auino, 2023-01-18:
Everything was correct.
The only thing different was the `TokenSecret`: as I generated it with the app on its previous version, it was not working properly/it was generated wrong.
Now I re-generated the `TokenSecret` and everything is working correctly.
So, in case someone is having the same issue, my suggestion is to restore the `TokenSecret` to the "plain text" version.

akopachov, 2023-01-18:
Glad it started working for you!