akopachov/flipper-zero_authenticator #67: Setup Guide

Kf637, 2023-01-18:
It would be nice to have a setup guide on the Flipper or just something on the README file on this GitHub.
I'm having issues where I don't know how to set it up and don't wanna have to ask someone how to set it up...

akopachov, 2023-01-18:
Well, there is some short description [here in FAQ](https://github.com/akopachov/flipper-zero_authenticator/blob/v1.6.0/FAQ.md#how-to-get-it) regarding how to get this app running on Flipper. Then initial setup is fairly simple, app just asks you whether you want to set PIN or not, I didn't feel that it is something what need to be explained.

The only thing what might be covered better is some example of how to add new token to the app, here I agree and I'll try to add it.

What exact problems you have during setting things up?

Kf637, 2023-01-18:
I don't see a section where it says how I can set it up.

akopachov, 2023-01-18:
There is section called "How to get it?" where it is explained that you need to put FAP file on your Flipper. That's it, it is enough to run it on your flipper. After you run it first time app will walk you through setting up PIN if you would like.

What other setup instructions are you missing? I need to understand where you had trouble to know what to explain

Kf637, 2023-01-18:
It's already installed, I don't know how to add the one time codes due to there being multiple hashing methods.

akopachov, 2023-01-18:
oh, I see now.
Agree, it probably makes sense to add some sample walkthrough on how to add new token for some popular provider.

Just in case if you don't know which hashing method to use - use SHA1, as it is used by majority of providers