akopachov/flipper-zero_authenticator #68: Load new secret token through file

Kf637, 2023-01-19:
It would be nice to be able to add a key from a file.

Exapmle:
1. Open Notepad and paste the following text
`Filetype: Token`
`Version: 0`
`Secret: Code1234`
`Algorithm : SHA1`
2. Replace 'Secret' and 'Algorithm' with your secret token and the hashing algorithm
3. Save the file as a '<Filename>.tokensecret'
4. Move the file to the Flipper Zero
5. Open the Flipper Authenticator application and press 'Load New Token from File'
6. Find the `.tokensecret` file and load it
7. The application will add the Secret automaticly and delete the `<Filename>.tokensecret`

This would be great for users who has multiple keys to tranfer to their Flipper and lowers the risk of error.

akopachov, 2023-01-20:
Hey, I don't think it make that much of a sense for users, give that they can achieve the same in a two existing ways:
1. Edit manually `/ext/authenticator/totp.conf` file and add all the necessary tokens there, then once application will be launched it will notice that there are unencrypted plain tokens and will encrypt them automatically.
2. Use CLI command `totp new` to add new token - this is much more userfriendly and secure way of adding tokens compared to manipulating files.

Kf637, 2023-01-20:
Didn't know that it was possible, thanks for the awnser.

closed 2023-01-20 by Kf637