akopachov/flipper-zero_authenticator #69: Digital signature verification failed

Kf637, 2023-01-22:
I'm having an issue after updating my firmware, not sure why this is happening.
[image]

akopachov, 2023-01-23:
Could you please let me know what firmware are you using and what is the version of it?

I just tried latest [v1.6.0](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.6.0) on [latest Unleashed firmware](https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-025) and it works just fine.

Kf637, 2023-01-23:
Rogue Master Custom Release RM0123-0236-0.75.1-c8d2036

akopachov, 2023-01-23:
I can't find this release in RM's github [here](https://github.com/RogueMaster/flipperzero-firmware-wPlugins/tags).
If this is some really custom build, it is hard for me to guess what could be wrong as it could be just some unstable or incompatible firmware.

Probably it might be better for you to contact the owner of the custom RM release you have.

Kf637, 2023-01-23:
Here's the build. [RM0123-0236.zip](https://github.com/akopachov/flipper-zero_authenticator/files/10479135/RM0123-0236-KF637Build.zip)

akopachov, 2023-01-23:
How did you get this build? If it is not on the RM's release page, then probably it is something unofficial and I assume may have some issues in it.

Kf637, 2023-01-23:
> How did you get this build? If it is not on the RM's release page, then probably it is something unofficial and I assume may have some issues in it.

It's a custom patron build

akopachov, 2023-01-23:
Based on the message - looks like you do not have PIN setup, right? Was it not set all the time or have you removed PIN?

If so, then your device UID is used instead of PIN to encrypt secrets. And just in case if app was not able to decrypt the data using device UID it will assume that it is wrong device and will show error message like on your screenshot.

Kf637, 2023-01-23:
I don't have a pin set and never had.

akopachov, 2023-01-23:
okay, then if you didn't change your flipper and didn't manually corrupt `totp.conf` file and the only thing what lead to this issue is firmware update then somehow it happened that `furi_hal_version_uid` function in this firmware returns information which is differ with previous firmware you are running.

This should never happen because `furi_hal_version_uid` returns unique device identifier which should stay static regardless of firmware. But as we can see, somehow that happened that.

Have you tried to flash another firmware and run there?

Kf637, 2023-01-23:
> okay, then if you didn't change your flipper and didn't manually corrupt `totp.conf` file and the only thing what lead to this issue is firmware update then somehow it happened that [...]

His old firmware version works.

akopachov, 2023-01-23:
Then I guess it worth it to raise issue in his backlog. I don't think anything could be done on my side

closed 2023-01-23 by akopachov