akopachov/flipper-zero_authenticator #82: 与Authy软件同步

metr0man, 2023-02-28:
How do I synchronize with the Authy installed on other devices? I got the authy token from localhost in local debugging and the value was null, and I'm not sure what token name is filling in. help plz!

metr0man, 2023-02-28:
https://gist.github.com/mkyrychenko/6d3d2bcfdafeaeaae09ae81eb285af90
https://gist.github.com/gboudreau/94bb0c11a6209c82418d01a59d958c93

closed 2023-02-28 by metr0man

akopachov, 2023-03-01:
Glad you found solution

akopachov, 2023-03-01:
Added the approaches you posted to [official wiki](https://github.com/akopachov/flipper-zero_authenticator/wiki/How-to-migrate-from-another-authenticator-app%3F) so that other people would know what to do if they face the same situation as you.. Thank you for providing that links