akopachov/flipper-zero_authenticator #92: API version mismatch

yasbean, 2023-03-21:
Is it just me?  Compiling this update results in "Preload failed: API version mismatch."  Trying the pre-compiled versions lead to the same error.  I tried re-cloning the firmware, but still no luck.  I am on the latest FW, 0.79.1.

akopachov, 2023-03-21:
Which precompiled file have you used?

Have you tried to update firmware submodules using this command `git submodule update --remote --force` and the compiling it manually?

yasbean, 2023-03-21:
I first tried compiling totp manually.  I then tried grabbing totp_official-stable_fw.fap.  I then tried deleting the entire firmware directory, cloning from fresh, and compiling manually.  Same result.  Very odd.
I don't know if this helps but I have a soft link "ln -s applications_use/flipper-zero_authenticator/totp applications_user_totp, so when I run ./ftb fap_totp, I receive the following:
```
fbt: warning: Folder /home/cds/source/flipperzero-firmware/applications_user/flipper-zero_authenticator: manifest application.fam is missing
LoadAppManifest, line 32, in file "/home/cds/source/flipperzero-firmware/scripts/fbt_tools/fbt_apps.py"
        SDKCHK  firmware/targets/f7/api_symbols.csv
API version 19.0 is up to date
        APPCHK  build/f7-firmware-D/.extapps/totp.fap
```

akopachov, 2023-03-21:
I do not remember exactly why but I moved out of making soft links when building this TOTP app. Instead I'm using [this trick](https://github.com/akopachov/flipper-zero_authenticator/blob/master/sonar-build).

```bash
pushd flipperzero-firmware_official_dev
rm -rf applications/plugins/totp
sed -i 's/applications_user/../' site_scons/commandline.scons
./fbt fap_totp
popd
```

Not really sure if it is going to help or not, but you definitely may try. Also I'd highly suggest you to run `git submodule update --remote --force` instead of dropping firmware folders as this command will force git to pull latest stuff from the remote, and not using any locally cached stuff

yasbean, 2023-03-21:
I guess my git-fu is lacking.  I just use `git pull` to update my modules.   I did try `git submodule update --remote --force` as you suggested.
BTW, the trick may need to be updated since it looks like "external"has replaced "plugins" but I keep my manual plugins in applications_user.

akopachov, 2023-03-22:
Updated firmware submodules and built.
Could you try [v1.7.1](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.7.1)?

yasbean, 2023-03-22:
That did not help my compile issue, but the pre-built  `totp_official-stable_fw.fap` no works for me.  I will try compiling on my work computer in a few days and report back.

yasbean, 2023-03-28:
I successfully built totp v1.7.1 on firmware 0.80.1.

Thank you!

closed 2023-03-28 by yasbean