akopachov/flipper-zero_authenticator #96: BadBT doesn't work after running the app for the second time

zverev-iv, 2023-03-28:
I've used BadBT option on Windows 11. It works only if I pair PC and use badBT in the same app session.
Steps for reproduce:

0. Option BadBT previously enabled
1.  Open App, add "Control %flippername%"  device in windows bt options
2. Use BadBT (works nice)
3. Close App
4. Open App Again ("Control %flippername%"connects to PC)
5. Use BadBT Again (Fail)

P.S. also I tested it on my android, the app behaves similarly

akopachov, 2023-03-29:
Do you experience the same issue in bluetooth keyboard app?

eried, 2023-03-29:
not me, it always work, but the "connecting" screen sometimes is almost instant and sometimes is long

akopachov, 2023-03-29:
Works just fine with my Pixel 7 Pro.

Is there a chance somebody can give it a try with Linux?

zverev-iv, 2023-03-29:
I use these case for 2 apps;

0. Windows/Android don't synced with flipper
1. open app and sync flipper + first app use
2. close app + open app then + second app use
3. same as 2

firmware: unlshd 039 (both apps in versions from this firmware + extra pack)

for Bluetooth Remote app (Windows/Android):
0. ok
1. ok
2. ok
3. ok

for Authentificator (Windows/Android):
0. ok
1. ok
2. fail
3. fail

eried, 2023-03-29:
ps: I tried playing with the delays in your code, instead of 500 ms waiting used 5000 and 10000, but it is the same behaviour. Since is quite different to the Bluetooth Remote app (everything is 2 times wrapped in their code as "hid*"), it will require more research to find why is different.

akopachov, 2023-03-29:
Could you please try with [this](https://file.io/NSuQlWORq1ay) FAP?

eried, 2023-03-29:
> Could you please try with [this](https://file.io/NSuQlWORq1ay) FAP?

same, first time ok. After, just connects but for a moment.

PS: nice icons!

akopachov, 2023-03-29:
So, I'm kinda lack of ideas. Especially given the fact that it behaves that much different and a bit random (at least on my side).

Will try to find other laptop running windows to have some more test environments

eried, 2023-03-29:
Also, if you press fast left or right while the app is loading the "BT icon" it gets frozen

akopachov, 2023-03-30:
Could you guys try [v1.8.0](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.8.0). Just for me to understand if issue still persists in latest release

eried, 2023-03-30:
Hey, I think it works ok now. It was not working as before, I unpaired and now I cannot make it fail anymore.

In one of the tests took 6 or more seconds, but that is expected as with the BT keyboard

akopachov, 2023-03-30:
Glad to hear it acts better! Thank you for all  the testing and helpful feedback

akopachov, 2023-03-31:
I guess this issue could be closed now

closed 2023-03-31 by akopachov