akopachov/flipper-zero_authenticator #111: Steam Guard Support

FLozyXD, 2023-04-02:
Hi!
Is there any way of adding steam guard support? It is avaliable in Authenticator Pro, adding sectrets through Steam Desktop Authenticator. Is it possible to add a steam algorithm and 5 digits length?

akopachov, 2023-04-03:
I think so.

Adding 5 digits is super-easy thing to do, and as to steam algo - let me see what can I do on this.

Thanks for reporting this service to me

closed 2023-04-05 by akopachov

akopachov, 2023-04-05:
Should be implemented in [v1.8.4](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.8.4)

FLozyXD, 2023-04-05:
How do i use the steam algo? I can choose only between sha1, sha256 and sha512. 5 digits option is available btw

akopachov, 2023-04-05:
Steam algo is available to add via CLI. Also I'm merging the fix to add it to "Add new token" screen now, but I doubt you will be using this way as it is obsolete and kinda annoying.

akopachov, 2023-04-05:
[v1.8.5](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.8.5)

[Here is draft version of FAQ page](https://github.com/akopachov/flipper-zero_authenticator/wiki/Does-it-support-Steam-style-tokens%3F)