akopachov/flipper-zero_authenticator #118: error: implicit declaration of function 'furi_hal_bt_get_profile_adv_name'  (1.8.5, fw official 0.80.1)

yasbean, 2023-04-05:
Since 1.8.0, when compiling on official fw, receiving the following errors: 
```
	AR	build/f7-firmware-D/.extapps/totp/libroll_value.a
	RANLIB	build/f7-firmware-D/.extapps/totp/libroll_value.a
applications_user/totp/workers/bt_type_code/bt_type_code.c: In function 'totp_bt_type_code_worker_init':
applications_user/totp/workers/bt_type_code/bt_type_code.c:143:9: error: implicit declaration of function 'furi_hal_bt_get_profile_adv_name' [-Werror=implicit-function-declaration]
  143 |         furi_hal_bt_get_profile_adv_name(FuriHalBtProfileHidKeyboard),
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
applications_user/totp/workers/bt_type_code/bt_type_code.c:143:9: error: passing argument 2 of 'memcpy' makes pointer from integer without a cast [-Werror=int-conversion]
  143 |         furi_hal_bt_get_profile_adv_name(FuriHalBtProfileHidKeyboard),
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         int
In file included from ./furi/core/memmgr.h:10,
                 from ./furi/furi.h:10,
                 from applications_user/totp/workers/bt_type_code/bt_type_code.h:4,
                 from applications_user/totp/workers/bt_type_code/bt_type_code.c:1:
/home/cds/source/flipperzero-firmware/toolchain/x86_64-linux/arm-none-eabi/include/string.h:31:35: note: expected 'const void * restrict' but argument is of type 'int'
   31 | void *  memcpy (void *__restrict, const void *__restrict, size_t);
      |                                   ^
applications_user/totp/workers/bt_type_code/bt_type_code.c:147:9: error: implicit declaration of function 'furi_hal_bt_get_profile_mac_addr' [-Werror=implicit-function-declaration]
  147 |         furi_hal_bt_get_profile_mac_addr(FuriHalBtProfileHidKeyboard),
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
applications_user/totp/workers/bt_type_code/bt_type_code.c:147:9: error: passing argument 2 of 'memcpy' makes pointer from integer without a cast [-Werror=int-conversion]
  147 |         furi_hal_bt_get_profile_mac_addr(FuriHalBtProfileHidKeyboard),
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         |
      |         int
In file included from ./furi/core/memmgr.h:10,
                 from ./furi/furi.h:10,
                 from applications_user/totp/workers/bt_type_code/bt_type_code.h:4,
                 from applications_user/totp/workers/bt_type_code/bt_type_code.c:1:
/home/cds/source/flipperzero-firmware/toolchain/x86_64-linux/arm-none-eabi/include/string.h:31:35: note: expected 'const void * restrict' but argument is of type 'int'
   31 | void *  memcpy (void *__restrict, const void *__restrict, size_t);
      |                                   ^
applications_user/totp/workers/bt_type_code/bt_type_code.c:151:5: error: implicit declaration of function 'furi_hal_bt_set_profile_adv_name' [-Werror=implicit-function-declaration]
  151 |     furi_hal_bt_set_profile_adv_name(FuriHalBtProfileHidKeyboard, new_name);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
applications_user/totp/workers/bt_type_code/bt_type_code.c:152:5: error: implicit declaration of function 'furi_hal_bt_set_profile_mac_addr' [-Werror=implicit-function-declaration]
  152 |     furi_hal_bt_set_profile_mac_addr(FuriHalBtProfileHidKeyboard, context->bt_mac);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
applications_user/totp/workers/bt_type_code/bt_type_code.c:162:5: error: implicit declaration of function 'bt_enable_peer_key_update' [-Werror=implicit-function-declaration]
  162 |     bt_enable_peer_key_update(context->bt);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
	SDKCHK	firmware/targets/f7/api_symbols.csv
scons: *** [build/f7-firmware-D/.extapps/totp/workers/bt_type_code/bt_type_code.o] Error 1
API version 20.0 is up to date

********** FBT ERRORS **********
build/f7-firmware-D/.extapps/totp/workers/bt_type_code/bt_type_code.o: Error 1

```

yasbean, 2023-04-05:
BTW:  totp_official-stable_fw.fap 1.8.5 also API version mismatch on official fw 0.80.1

closed 2023-04-06 by akopachov

akopachov, 2023-04-06:
[v1.8.6](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.8.6) should be compatible with latest firmware (0.80.1).

As to build issue you mentioned: starting from version 1.8.0 I added support of [Xtreme](https://github.com/ClaraCrazy/Flipper-Xtreme) firmware and its specific features. If you are running `build.ps1` to build all, then you should not see any issue as that file will take care about everything. 

Just in case if you are building authenticator somehow else, then you should manually update [totp/features_config.h](https://github.com/akopachov/flipper-zero_authenticator/blob/master/totp/features_config.h) file by changing this line: `#define TOTP_TARGET_FIRMWARE TOTP_FIRMWARE_XTREME` to `#define TOTP_TARGET_FIRMWARE TOTP_FIRMWARE_OFFICIAL_STABLE ` (or `#define TOTP_TARGET_FIRMWARE TOTP_FIRMWARE_OFFICIAL_DEV`). This change will make code compatible with appropriate firmware and you should no longer see any compiler issues.

akopachov, 2023-04-06:
Updated [wiki page](https://github.com/akopachov/flipper-zero_authenticator/wiki/How-to-build-it%3F) with this info