akopachov/flipper-zero_authenticator #122: setup pin code leads to incorrect code generation

zverev-iv, 2023-04-07:
Case: I Have

akopachov, 2023-04-07:
Could you please provide Authenticator version and your Flipper's firmware and its version?

zverev-iv, 2023-04-07:
update information unlshd-042e- https://github.com/DarkFlippers/unleashed-firmware/releases/tag/unlshd-042

akopachov, 2023-04-07:
Here is what I did to test this on my Flipper:
1. Grab new secret [here](https://www.verifyr.com/en/otp/check#totp)
2. Added it to the app
3. Made sure generated code is correct
4. Removed PIN via `totp pin remove`
5. Made sure generated code is correct
6. Restarted app
7. Made sure generated code is correct
8. Setup new PIN via `totp pin set`
9. Made sure generated code is correct
10. Restarted app
11. Made sure generated code is correct

So far was not able to reproduce the issue.

Do you remember if any unusual things happened? Or maybe something what is not covered by the test steps I did above? Can you try to reproduce this issue using test website I provided above?

akopachov, 2023-04-07:
In any case, logged #123 to make sure there is backup

akopachov, 2023-04-11:
> Do you remember if any unusual things happened? Or maybe something what is not covered by the test steps I did above? Can you try to reproduce this issue using test website I provided above?

Is there any chance to try it?

akopachov, 2023-04-12:
Closing as I was not able to reproduce this issue, author is not responding and the necessary improvements to take a backup before setting\removing PIN is added.

closed 2023-04-12 by akopachov

zverev-iv, 2023-04-13:
Hi! Sorry for late answer.
I can't reproduce it.
I tried many times. actually happened **twice**, but perhaps this is due to the time shift in rtc (for example, it leads to a firmware update or something else)