akopachov/flipper-zero_authenticator #134: API mismatch

yasbean, 2023-04-13:
**Describe the bug**
I am again getting API mismatch when compiling 1.9.1 for OFFICIAL_STABLE. fw 0.81.1

**To Reproduce**
1. Edit topt/features.h to replace  #define TOTP_TARGET_FIRMWARE TOTP_FIRMWARE_XTREME with
#define TOTP_TARGET_FIRMWARE TOTP_FIRMWARE_OFFICIAL_STABLE
2. ./fbt fap_totp
3. Copy totp.fap to SD Card/Apps/Tools/

**Expected behavior**
Have an icon next to Applications/Tools/Authenticator and be able to run the application

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Device information (please complete the following information):**
 - Firmware: [Official stable]
 - Firmware version [0.81.1]

**Additional context**
Add any other context about the problem here.

akopachov, 2023-04-13:
Not really sure why the build artifacts you are producing are not working for you. If you are sure that to build you are suing the same firmware version as you are running on your flipper - it should work.

In any case, will publish new version built based of latest firmware soon

akopachov, 2023-04-13:
Here it is - [v1.9.2](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v1.9.2)

closed 2023-04-13 by akopachov

yasbean, 2023-04-13:
Very odd, but this works.

Thank you!