akopachov/flipper-zero_authenticator #136: [BUG] Authenticator does not load

satanasdiabolo, 2023-04-14:
**Describe the bug**
When starting my Authenticator application, I only get the rotating hourglass, and my Flipper is freezed

**To Reproduce**
Start the application

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Device information (please complete the following information):**
 - Firmware: Unleashed
 - Firmware version unlshd-043e

akopachov, 2023-04-14:
Are you using Flipper Authenticator builtin into firmware or have you downloaded and installed it manually from this repo?

satanasdiabolo, 2023-04-14:
I'm using the version provided with unleashed firmware

akopachov, 2023-04-14:
Try to use the latest from this repo.

[v1.9.2 - totp_official-dev_unleashed_fw.fap](https://github.com/akopachov/flipper-zero_authenticator/releases/download/v1.9.2/totp_official-dev_unleashed_fw.fap)

satanasdiabolo, 2023-04-14:
Thank you very much, it's working fine now!