akopachov/flipper-zero_authenticator #146: BadUSB/BT capslock activated (german layout)

philipneill, 2023-04-28:
First of all, thank you for this very nice tool!

The Bug:
If you use a german keyboard layout in windows, capslock is activated an then its outputing all those special characters, instead of the numbers.

To reproduce, just select a german keyboard layout and try.

akopachov, 2023-04-29:
Thanks for reporting this!

I doubt I'd be able to find this bug myself given that I never use german keyboard layout 😄

philipneill, 2023-04-29:
You dont need to use a physical layout, just change the windows keyboard to a german one and the issue will happen.

https://support.microsoft.com/en-us/windows/manage-the-input-and-display-language-settings-in-windows-12a10cb4-8626-9b77-0ccb-5013e0c7c7a2#:~:text=the%20following%20methods.-,Change%20your%20keyboard%20layout,you%20have%20only%20one%20installed.)

closed 2023-04-29 by akopachov

akopachov, 2023-04-29:
Thanks again for all the input. It allowed me to make this part of code easier and better than it was.

This should be fixed in [v2.0.3](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v2.0.3)

philipneill, 2023-04-29:
Thank you so much!!

PR #147 (2023-04-29): Improved token input automation code to get rid of caps lock key usage
* Fixed #146
* Updated firmware submodules