akopachov/flipper-zero_authenticator #149: [BUG] Steam Guard seems to be broken

FLozyXD, 2023-05-21:
**Describe the bug**
I've just recovered secrets using the latest guide and Flipper shows the wrong code. However, by using this secret in Authenticator Pro I can get right codes.

**To Reproduce**
Just follow the latest guide to get secrets and add them via Flipper CLI

**Expected behavior**
It is expected to get correct Steam Guard codes from Flipper.

**Device information (please complete the following information):**
 - Firmware: Unleashed Firmware
 - Firmware version: dev-585e

**Additional context**
By the way, any other auth codes work fine

akopachov, 2023-05-21:
Hey, just to make sure - did the secret token you extracted from steam looked like Base64-encoded? (alpha-numeric chars in different case and possible `/` and/or `=` symbols?

FLozyXD, 2023-05-21:
Only letters and numbers, no other symbols. Took it from this string: "uri":"otpauth:\/\/totp\/Steam:[account_name]?secret=[SECRET]&issuer=Steam"

akopachov, 2023-05-21:
Were those alphanumeric symbols all in the same case?

FLozyXD, 2023-05-21:
Yep, all the letters were uppercase

akopachov, 2023-05-21:
Then it looks like it is Base32-encoded, which is a bit weird for Steam secrets. Anyway, try update CLI command by changing `-e base64` to `-e base32`.

FLozyXD, 2023-05-21:
Well... -e base32 works fine, bruh. Maybe add this to FAQ?

akopachov, 2023-05-21:
Yeah, will definitely update wiki according to this finding (I never heard before about non-base64 encoded steam secrets). Thanks for reporting it!

akopachov, 2023-05-22:
Updated [appropriate wiki section](https://github.com/akopachov/flipper-zero_authenticator/wiki/Does-it-support-Steam-style-tokens%3F) to better explain `-e base64` usage.

closed 2023-05-22 by akopachov