akopachov/flipper-zero_authenticator #153: [BUG] API mismatch and signature failure.

Anatejl, 2023-06-03:
**Describe the bug**

After update for XFW48, it rolled out latest version, it returned errors, and didn't work. 
Then I reinstalled the app from repo manually, errors persist.

**To Reproduce**

1. Go to apps menu
2. Select authenticator
3. See error

**Expected behavior**

Regular behavior for the app 

**Screenshots**

[image]

[image]

**Device information (please complete the following information):**
 - Firmware: XFW48
 - App - 2.1.1

**Additional context**
Add any other context about the problem here.

ClaraCrazy, 2023-06-04:
api mismatch is just bc you installed it yourself, instead of using the fap provided. 

as for the digital verificatuon thingy, idk. thats an app specific thing. Have fun @akopachov

ClaraCrazy, 2023-06-04:
Seems to me that [this](https://github.com/akopachov/flipper-zero_authenticator/blob/3538f31fe9cb051de9df0d3b0e9d2e7252da7810/totp/services/crypto/crypto.c#L120) function returns false for some reason. After trying to open my, luckily empty, authenticator app, I could confirm this issue. I havent had tea yet so im not quite sure how it works, but it seems fairly easy to fix. Although it might be slightly harder to securely fix it, as this is a security function

akopachov, 2023-06-04:
Discussed that with @ClaraCrazy in DM. Looks like you changed flipper name which resulted to changing your flipper's UID and MAC. And the thing is that Authenticator app relies on UID when user is not setup PIN. We will try to find something better to rely on if possible. If not, then I guess there will not be any good choice other than 
1. Leave things as-is
2. Remove PIN-less authentication and force user to use PIN

Stay tuned...

akopachov, 2023-06-05:
Should be fixed in [v2.2.0](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v2.2.0)

closed 2023-06-05 by akopachov