akopachov/flipper-zero_authenticator #167: [FEATURE] Allow user to pick encryption key slot

akopachov, 2023-07-24:
Given [this](https://github.com/flipperdevices/flipperzero-firmware/issues/2885#issuecomment-1646664666) explanation I consider any key stored in slot number less than 5 as compromised. It is necessary to:

1. Implement migration to automatically migrate existing conf file to another default key slot
2. Let user change key slots via CLI

closed 2023-07-26 by akopachov

akopachov, 2023-07-26:
`totp pin` command got updated to accept additional `-c <slot>` argument, allowing user to pick which key slot to use

skotopes, 2023-07-26:
Explanation by @Skorp was incorrect, please see details in my comment https://github.com/flipperdevices/flipperzero-firmware/issues/2885#issuecomment-1652148724

akopachov, 2023-07-26:
Thanks for explanation. I already made appropriate changes in v3.0.1 (#169)