akopachov/flipper-zero_authenticator #171: [FEATURE] Use ALT CODES for universal keyboard layouts handling

Shotman, 2023-07-26:
I'm using the french keyboard layouts and my numbers are not the default characters on in, so I have to engage capslock on my physical keyboard to enter the TOTP

I've seen in badUSB and badBT payloads that it's possible to use of ALT CODES to avoid everything regarding that

https://flippermaker.github.io allows you to generate ALT CODES for a given text : 
ALTCHAR 048
ALTCHAR 049
ALTCHAR 050
ALTCHAR 051
ALTCHAR 052
ALTCHAR 053
ALTCHAR 054
ALTCHAR 055
ALTCHAR 056
ALTCHAR 057
here are the ones for digits 0 to 9 
https://github.com/d3usw4re/AltCoder and this repo does some alt code conversion for bad usb on arduino

akopachov, 2023-07-27:
Good idea! Will definitely try

akopachov, 2023-07-27:
What version for Authenticator are you using now? This issue sounds very similar to #146 which was fixed in v2.0.3 and all the above versions.

akopachov, 2023-07-27:
Oh, I see. So basically, on your keyboard layout to type digits you need to hold "shift" key pressed, am I correct?

Shotman, 2023-07-27:
Exactly and I think since authenticator just uses the digits it would be wise to switch to alt codes to handle ANY layout universally

Shotman, 2023-07-28:
I just booted my flipper and both badUSB and badBT (for ducky scripts etc) apps (on the unleashd) on it allow to choose keyboard layout through a config screen, so might be quite easy to reimplement what they did

closed 2023-07-31 by akopachov