akopachov/flipper-zero_authenticator #186: [FEATURE] : Possibility to scan qrCode for import / Possibility to pass uri to parse

noudin-ledger, 2023-08-09:
**Is your feature request related to a problem? Please describe.**
Typing keys is painful and even with CLI, some websites only provide a qrCode.

**Describe the solution you'd like**
If we were able to give the qrCode file or the google-auth format uri it would save a lot of time and automatically fill the right fields

**Describe alternatives you've considered**
Client side automation of parsing to submit the right CLI command to totp cli

**Additional context**

URI format : https://github.com/google/google-authenticator/wiki/Key-Uri-Format

akopachov, 2023-08-09:
Hey, thanks for a good idea. I doubt I'd be able to parse QR code, as website may show it using different formats (gif, png, svg and etc.) and add parser for each format is going to be a hell. However it should be absolutely doable to parse `otpauth://...` URI.

akopachov, 2023-08-14:
Just FYI: this feature request is not abandoned or so.

Initially I thought about implementing parsing TOTP URI right inside application, to let user copy&paste straight URI. But then I realized that there is a better approach. It is way more complicated, but I believe it will be way more useful to a users. For now code name for this approach is "Flipper Authenticator Companion App" which is going to be cross-platform desktop app allowing you to interact with TOTP application in a way more userfriendly way, including scanning QR codes automatically by the app.

For now this approach is in early development, but I definitely make it public once first POC will be ready.

noudin-ledger, 2023-08-14:
Ooooh this is even better indeed and that's a great idea!
I had not considered it abandoned or so, I know quite a lot about the time it takes and that we have our lives to handle and everything apart from building nice softwares.
I'm no C expert but if you need some help I can try to find some time and upgrade my C to help you!

akopachov, 2023-08-25:
Finally, FAC (Flipper Authenticator Companion) app is there!
https://github.com/akopachov/flipper-zero_authenticator-companion

Checkout latest release and feel free to post any ideas\issues to its repo.

Closing this issue as it is done :)

closed 2023-08-25 by akopachov

noudin-ledger, 2023-08-29:
Good I'll have a look and I'll test it !