akopachov/flipper-zero_authenticator #192: [Question] Same answer as gnome authenticator?

FalcoGer, 2023-08-25:
Is it possible to import the data from another authenticator app, such as gnome authenticator? I still have the secrets/urls I entered into gnome authenticator, but no matter what setting I choose for flipper's authenticator, it's always different from what gnome authenticator spits out. I tried all the hashing algorithsm and even added the timezone offset forward and backwards, but it just doesn't want to agree.

akopachov, 2023-08-25:
It is possible to use Flipper Authenticator is you have plain token secret. If you have URI - that is also suitable, but **out of that URI you need to pick only secret piece**. 
Here is an example: 
[image]

Then use [CLI](https://github.com/akopachov/flipper-zero_authenticator/wiki/Is-there-a-CLI%3F) to add this secret to the app. Just in case if generated token doesn't seems to be valid there could be [multiple reasons](https://github.com/akopachov/flipper-zero_authenticator/wiki/Flipper-Authenticator-generates-invalid-codes%2C-why-so%3F) why:
- Clock on device is not precise
- Timezone is not correct
- Token secret is not correct
- Token hashing algo is not correct

Usually it is either clock or timezone, but also worth to make sure token secret is correct too.

Related topics:
[How to migrate from another authenticator app?](https://github.com/akopachov/flipper-zero_authenticator/wiki/How-to-migrate-from-another-authenticator-app%3F)
[Flipper Authenticator generates invalid codes, why so?](https://github.com/akopachov/flipper-zero_authenticator/wiki/Flipper-Authenticator-generates-invalid-codes%2C-why-so%3F)
[Is there a CLI?](https://github.com/akopachov/flipper-zero_authenticator/wiki/Is-there-a-CLI%3F)

akopachov, 2023-08-25:
Also, if you have original QR codes - you may use [companion app](https://github.com/akopachov/flipper-zero_authenticator-companion) to quickly add it to Authenticator app

FalcoGer, 2023-08-25:
Thank you. Having the CLI really helped debug the issue. It was a timezone problem, but also for some reason the desktop application is about 20 seconds out of sync with the clock, despite the clocks being exactly the same (synced with qFlipper just moments before).

FalcoGer, 2023-08-25:
Is it possible to have it switch timezones automatically with daylight savings?

akopachov, 2023-08-25:
It is not possible as flipper is not aware of all different timezones and their dst rules. Implement such a database for flipper would overkill it :)
However I'll try to solve this in companion app for desktop to let it sync datetime and timezone at every run

FalcoGer, 2023-08-25:
Flipper is aware of the date. I was thinking perhaps simply of a setting that would check that date and time and a flag and toggle between summer and winter time, then store that updated flag to storage. But maybe that is something that should be done at a higher level (flipper OS)?

FalcoGer, 2023-08-25:
I will close this and open a new issue since it's off topic.

closed 2023-08-25 by FalcoGer