akopachov/flipper-zero_authenticator #197: not working!! help! [BUG]

grugnoymeme, 2023-09-02:
Hello. Everytime, after put the name, when i click save after put the secret token too, it re-send me to the main screen wher i am asked: press to add new token... i tryed with github, google, instagram, and no one is working.. i tryed to write all UPPERCASE or all lowercase or mixed,just like it was written but nothing.. can u help me please? im running xtreme (dev branch) at the moment of the test.

akopachov, 2023-09-03:
Highly suggest you to try [companion app](https://github.com/akopachov/flipper-zero_authenticator-companion). It is way easier and user friendly than any other method.

grugnoymeme, 2023-09-03:
yes i now, but i cannot understand why it's not working.. sometimes it accepts the secret token, but then generates a no-working totp..  do u have any idea about this?
thank u. very appreciate ur job!!

akopachov, 2023-09-03:
There is a [wiki page] (https://github.com/akopachov/flipper-zero_authenticator/wiki/Flipper-Authenticator-generates-invalid-codes%2C-why-so%3F) explaining why generated token might be incorrect.

Usually it is because of invalid timezone offset which you have to set in app settings (or via companion app) AND/OR inaccurate clock on your flipper device.

In companion app you may also set "auto sync" option for datetime and timezone, so that every sinhle time you connect flipper and launches companion app it will sync things up

grugnoymeme, 2023-09-03:
yes ok! i'll try i just wanted to understand if i can make it work also without the companion app.. cause i thought to set the correct timezone offset, and my flipper clock is set as well in the right way.. in fact, sometimes i tryied i was taking back to the main screen, and sometimes instead i was permitted to save the token but it wasn't good. anyway: i'll read all the wiki page (i didn't noticed that, thank u!!, and i'll try the companion app, but it would be great to use just the flipper without the companion). Thank u. bye!!!

akopachov, 2023-09-03:
Yeah, absolutely. It can be used without companion app. That case CLI (also covered in wiki) is your choice, as it is more user friendly than adding token via flipper gui.

In any case, feel free to ping me if any questions/issues.

grugnoymeme, 2023-09-03:
super kind and polite!!! than u very much!!!

grugnoymeme, 2023-09-03:
i'll try definetly the CLI way!

closed 2023-09-11 by akopachov