akopachov/flipper-zero_authenticator #202: [BUG] Google Authenticator Tokens not working

Ay1tsMe, 2023-09-15:
**Describe the bug**
I was able to import my google authenticator tokens into the flipper zero authenticator. However, when i tried testing them on various websites, it says its incorrect. Im currently using the authenticator app built into the xtreme firmware. Is this error happening because this software needs to be updated so it works correctly with the companion?

**To Reproduce**
Steps to reproduce the behaviour:
1. Import google authenticator tokens
2. test them out and they dont work

akopachov, 2023-09-15:
The app which is bundled into Xtreme should work just fine with companion. There are [multiple reasons](https://github.com/akopachov/flipper-zero_authenticator/wiki/Flipper-Authenticator-generates-invalid-codes%2C-why-so%3F) why generated codes might be invalid. Usually it is inaccurate time on flipper device or/and invalid timezone set in app settings. You may fix both of this using companion app, by going to "settings" and turning on "date time" and "timezone" synchronization.

Ay1tsMe, 2023-09-16:
Adjusting the timezone offset fixed the issue. Thank you!

closed 2023-09-16 by Ay1tsMe