akopachov/flipper-zero_authenticator #207: [FEATURE] Get rid of OFW catalog specific stuff

akopachov, 2023-09-28:
I'm no longer going to publish any updates to OFW catalog until they stop their censorship and dictatorship practices (will add more details later). Given this - there is no sense to have anything what I added just to match OFW catalog "requirements".

closed 2023-09-28 by akopachov

RogueMaster, 2023-09-28:
OFW has horrible censorship practices but also, the discord server you linked is the most toxic in the community... so there are issues with both on my perspective...

akopachov, 2023-09-28:
Well, the discord server I linked might be not ideal. I never said it is the best server possible. But it is my choice to have exactly that link in app description, and as long as it does not violate any legal laws I do not see any reason why should I delete it.

Also, I checked their submission rules again, and was not able to figure out where exactly they have sort of "black-listed discord channel list" or so in it. So I guess it is just nothing more than just their wish to silently censor anything they don't like, shout up those people who are saying what they don't like, but not making it publicly to do not look like a jerks.

In any case, I'm not going to participate in this.

RogueMaster, 2023-09-28:
They are completely jerks, unfortunately, they are the jerks we are stuck with improving on this firmware at the moment. Hopefully for the better. In the future, I hope they get a team that isn't a censorship/propaganda machine to handle these types of things. They actually try to tell people it's illegal to have CFW. 🤦🏼‍♂️