akopachov/flipper-zero_authenticator #209: [BUG] Crashing on startup

UrSuLa360, 2023-10-11:
Hi, Authenticator app is crashing on start.
I think the issue is related to migration from old versions to latest.

Logs:
```
167890 [I][Loader] Loading /ext/apps/Tools/totp.fap
168004 [I][Elf] Total size of loaded sections: 37346
168007 [I][Loader] Loaded in 117ms
168010 [I][TOTP APP] App version: 5.5.0
168012 [I][TOTP APP] WolfSSL version: 5.6.3
168016 [I][AnimationManager] Unload animation 'L2_Dj_128x64'
168028 [I][TOTP APP] Obsolete config file version detected. Current version: 5;
Actual version: 10
168058 [I][TOTP APP] Took config file backup to /ext/apps_data/totp/backups/totp
.conf.20231011-8
168154 [I][TOTP APP] Applied migration to version 10
```

Steps to reproduce;
Install an old unleashed version i.e 056e, which contains and old version of Authenticator app and use the app.
Install unleashed version 064e, which contains a newer version of Authenticator app, then open authenticator app. It will crash

Here is the content of totp.conf, maybe it helps you.

```
Filetype: Flipper TOTP plugin config file
Version: 5
BaseIV: ED 68 2F C3 BC 31 F8 69 15 F0 2E 43 C6 8C B3 3B
Crypto: A5 D1 29 29 17 08 99 11 96 BE 19 35 1F 40 14 89
Timezone: 0.000000
PinIsSet: false
NotificationMethod: 3
```

akopachov, 2023-10-12:
Looks like the app version you are referring to is at least 2 major releases behind. In app version 5.x.x I've dropped support of crypto algorithms v1, which is used in the app version you are referring to. 

Looks like you do not have any tokens added in the app yet, so the easiest way to do is to just delete `/ext/apps_data/totp/totp.conf` file. This will reset the app to default, app will regenerate that file again using all the latest and greatest values.

UrSuLa360, 2023-10-12:
👍 Thanks for your response.
Deleting the file fixed the issue.

closed 2023-10-12 by UrSuLa360