akopachov/flipper-zero_authenticator #211: Building on Linux for Official FW

yasbean, 2023-10-25:
**Describe the bug**
I am trying to compile the latest version on linux but the totp/features_config.h has been moved. When I build, I receive an ApiTooNew error.  Where can I set the firmware version?

**Device information (please complete the following information):**
 - Firmware: Official stable
 - Firmware version 0.93.0

akopachov, 2023-10-26:
Starting from v5 I updated all the necessary code to use [uFBT](https://github.com/flipperdevices/flipperzero-ufbt). 

Once you install it, all you need to do to build for latest official stable FW is:

```bash
cd totp
ufbt update --channel=release
ufbt
```

I also setup ["Nightly build" action](https://github.com/akopachov/flipper-zero_authenticator/actions/workflows/nightly-build.yml) which once per day builds latest application version using latest firmwares, so you may download and use its artifacts too

Just in case if you need very specific version of official stable firmware, then it is also possible using uFBT, just take a look at their ["Managing the SDK"](https://github.com/flipperdevices/flipperzero-ufbt#managing-the-sdk) section

yasbean, 2023-10-26:
That has resolved it.  Thank you!

closed 2023-10-26 by yasbean