akopachov/flipper-zero_authenticator #212: what is the counter?

grugnoymeme, 2023-10-28:
hi what is the totp counter?? where i can find it? thank u!

akopachov, 2023-10-28:
Counter is needed for HOTP tokens only. Usually it is something what HOTP providers gives user together with secret. If it is not given by HOTP provider then highly-likely it is zero.

akopachov, 2023-10-28:
Emphasizing - counter is HOTP-only thing, and is not applicable for TOTP token types

grugnoymeme, 2023-10-28:
yes but flipper fap ask me the counter also for the totp...

akopachov, 2023-10-28:
It is asking for counter only if HOTP token type is picked. Make sure you picked correct token type (HOTP VS TOTP).

grugnoymeme, 2023-10-28:
yes.. it asking me for the totp.. idk...

akopachov, 2023-10-28:
What is the firmware you are using?

grugnoymeme, 2023-10-28:
unleashed

akopachov, 2023-10-28:
Let's try to use fap and fal files from latest [nightly build for unleashed] (https://github.com/akopachov/flipper-zero_authenticator/actions/workflows/nightly-build.yml) as I'm not so sure what exact version is merged to Unleashed

akopachov, 2023-10-28:
Nvm. Let me dig more. I think I know what could be wrong

closed 2023-10-30 by akopachov

akopachov, 2023-10-30:
Should be fixed in [v5.6.0](https://github.com/akopachov/flipper-zero_authenticator/releases/tag/v5.6.0)

grugnoymeme, 2023-10-30:
ok i'll try this evening

commit 20965dae (2023-10-30): Fixed #212 (#213)

* Fixed #212. Jusat a bit of refactoring

* CLang format changes

---------