akopachov/flipper-zero_authenticator #215: Usb input

mfulz, 2023-11-19:
Check my comment on this old issue: https://github.com/akopachov/flipper-zero_authenticator/issues/24

akopachov, 2023-11-20:
That's interesting.
What OS are you running?

mfulz, 2023-11-20:
Archlinux

akopachov, 2023-11-20:
Last time it was because host OS (windows 10 in that case) was not able to finish setting up keyboard before app actually starts sending key-press events. So given that there is no way for app to check whether host OS is actually ready or not, I just put a 0.5sec delay and that fixed that for Windows 10.

Is there anything fancy happening in Arch when it recognizes keyboard on one of the USB ports?

I guess we can make that delay configurable, to let users tune it the way they want to match their OS requirements.

mfulz, 2023-11-20:
No it looks fine, here is some dmesg output (including test tokens)

`[1935809.347317] hid-generic 0003:046D:C529.0097: input,hidraw11: USB HID v1.00 Keyboard [HID 046d:c529] on usb-0000:00:14.0-12.1/input0
8229[1935810.528639] usb 1-12.1: USB disconnect, device number 107
[1935811.090448] usb 1-12.1: new full-speed USB device number 108 using xhci_hcd
[1935811.182344] usb 1-12.1: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 1.00
[1935811.182348] usb 1-12.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1935811.182349] usb 1-12.1: Product: xxx 
[1935811.182350] usb 1-12.1: Manufacturer: Flipper Devices Inc.
[1935811.182351] usb 1-12.1: SerialNumber: flip_xxx
[1935811.190173] cdc_acm 1-12.1:1.0: ttyACM36: USB ACM device
[1935829.728587] usb 1-12.1: USB disconnect, device number 108
[1935830.450348] usb 1-12.1: new full-speed USB device number 109 using xhci_hcd
[1935830.544783] usb 1-12.1: New USB device found, idVendor=046d, idProduct=c529, bcdDevice= 1.00
[1935830.544787] usb 1-12.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[1935830.548645] input: HID 046d:c529 Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-12/1-12.1/1-12.1:1.0/0003:046D:C529.0098/input/input287
[1935830.603870] input: HID 046d:c529 Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-12/1-12.1/1-12.1:1.0/0003:046D:C529.0098/input/input288
[1935830.603944] hid-generic 0003:046D:C529.0098: input,hidraw11: USB HID v1.00 Keyboard [HID 046d:c529] on usb-0000:00:14.0-12.1/input0
193[1935831.776586] usb 1-12.1: USB disconnect, device number 109
[1935832.303693] usb 1-12.1: new full-speed USB device number 110 using xhci_hcd
[1935832.395744] usb 1-12.1: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 1.00
[1935832.395748] usb 1-12.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[1935832.395749] usb 1-12.1: Product: xxx
[1935832.395750] usb 1-12.1: Manufacturer: Flipper Devices Inc.
[1935832.395751] usb 1-12.1: SerialNumber: flip_xxx
`
You can see one time 8229 and second time 193

akopachov, 2023-11-20:
Let's try this build
[totp_xtreme_dev_fw.zip](https://github.com/akopachov/flipper-zero_authenticator/files/13411832/totp_xtreme_dev_fw.zip)
Here I changed initial delay from 0.5sec to 1sec

mfulz, 2023-11-20:
Which FW I can test this one? On the RM I'll get an API mismatch

mfulz, 2023-11-20:
ok non of these are working on RM (tried stable 93.1 and rc 95.1) will checkout on unleashed FW and give feedback

mfulz, 2023-11-20:
1sec delay is working flawlessly for me now:
- Different cables
- Different ports
- usb2
- usb3

no issues

akopachov, 2023-11-20:
Perfect. Then will add setting to let this delay be configurable.
Thanks for cooperation and reporting this issue.

closed 2023-11-20 by akopachov