akopachov/flipper-zero_authenticator #219: [FEATURE] Keys export

UndeRus, 2024-01-04:
**Is your feature request related to a problem? Please describe.**
I want to export keys from flipper to any external application

**Describe the solution you'd like**
Make ability to export keys from flipper zero

**Describe alternatives you've considered**
It may be implemented with keyboard output

closed 2024-01-08 by akopachov

akopachov, 2024-01-08:
Should be implemented in [v5.10.0](https://github.com/akopachov/flipper-zero_authenticator/releases/latest).

New CLI command `totp export` has been added. As a result of running this command an [URI list](https://github.com/google/google-authenticator/wiki/Key-Uri-Format) will be generated and printed to the console. This list should contain all the necessary information and be compatible with any authenticator allowing to import URI list files.

UndeRus, 2024-01-08:
Thank you very much for this feature! I already thought that I had lost my keys)