akopachov/flipper-zero_authenticator #221: Flipper Zero with latest official release will result in "API version mismatch" when installing from latest release (5.10.0)

besi, 2024-02-04:
I just got a new flipper zero and it is up to date with the latest official firmware

```
Version 0.98.2
Build date: 1.02.2024
Hardware: 12.F7****
Radio FW: 1.17.3 LITE
```

There are currently four open issues with regards to API version mismatch:
These people seem to be facing different issues than I am so this is why I'm opening a new issue here.

- #92 building from source
- #76 A year ago using a specific commit
- #16 using UNLSHD-012 (closed)
- #1 quite old

I did the following:

1. Plugin the flipper
2. Download `v5.10.0` [totp_official-stable_fw.zip](https://github.com/akopachov/flipper-zero_authenticator/releases/download/v5.10.0/totp_official-stable_fw.zip)
3. Extract all files onto the root SD card
4. Attempt to run `totp` which has a question mark for an icon
5. Error
     [image]
6. Try the same thing with [totp_unleashed_fw.zip](https://github.com/akopachov/flipper-zero_authenticator/releases/download/v5.10.0/totp_unleashed_fw.zip) since I read in one  of those issues that unleased works against the same API as the official firmware.
7. Same result :-(

[image]

I am not yet ready to attempt to build the project from source and I just wanted to give your app a spin so I do wonder how to run it on my new flipper running the latest official firmware...

Anyhow thanks for investing a lot of time and effort in something that's useful for a lot of people.

akopachov, 2024-02-04:
> @akopachov <https://github.com/akopachov>.
> <akopachov/flipper-zero_authenticator/issue/221/issue_event/11695122028@
> github.com>

You may try the artifact from nightly build -
https://github.com/akopachov/flipper-zero_authenticator/actions/workflows/nightly-build.yml
This action runs once a day building master branch over latest firmwares,
so highly - likely it should work for you.
If not - let me know and I'll build it manually.

closed 2024-02-05 by akopachov

besi, 2024-02-10:
@akopachov: I did try the latest version and I'm up and running now. Thanks!