akopachov/flipper-zero_authenticator #229: [BUG] Build fails for Official Firmware v 0.99.1

dengesCU, 2024-04-02:
**Describe the bug**
I am trying to build totp.fap for the latest official firmware 0.99.1 and it fails with the error message

**To Reproduce**
Steps to reproduce the behavior:
1. I cloned https://github.com/flipperdevices/flipperzero-firmware.git and placed the totp folder in the applications_user folder.
2. Running ./fbt fap_totp gives an error and the resulting totp.fap file does not work
3. See error
<pre><font color="#C4A000">LoadAppManifest, line 130, </font>in file &quot;/home/user/Documents/flipper/flipperzero-firmware/scripts/fbt_tools/fbt_apps.py&quot;
	CC	applications_user/totp/ui/scenes/authenticate/totp_scene_authenticate.c
	CC	applications_user/totp/ui/scenes/standby/standby.c
	SDKCHK	targets/f7/api_symbols.csv
	LINK	build/f7-firmware-D/.extapps/totp_d.elf
<font color="#4E9A06">API version 58.0 is up to date</font>
	APPMETA	build/f7-firmware-D/.extapps/totp.fap
	APPFILE	build/f7-firmware-D/.extapps/totp.fap
	FAP	build/f7-firmware-D/.extapps/totp.fap
	FASTFAP	build/f7-firmware-D/.extapps/totp.fap
	APPCHK	build/f7-firmware-D/.extapps/totp.fap
scons: *** [build/f7-firmware-D/.extapps/totp.impsyms] <font color="#FFCE51">build/f7-firmware-D/.extapps/totp.fap: app may not be runnable. Symbols not resolved using firmware&apos;s API: </font><font color="#AD7FA8">{&apos;I_WarningDolphinFlip_45x42&apos;}</font>

<font color="#F15D22"><b>********** FBT ERRORS **********</b></font>
<font color="#C4A000">build/f7-firmware-D/.extapps/totp.impsyms</font>: <font color="#FFCE51">build/f7-firmware-D/.extapps/totp.fap: app may not be runnable. Symbols not resolved using firmware&apos;s API: </font><font color="#AD7FA8">{&apos;I_WarningDolphinFlip_45x42&apos;}</font>
</pre>

**Preliminary fix**
The files totp/ui/scenes/authenticate/totp_scene_authenticate.c and totp/ui/scenes/standby/standby.c use the API symbol I_WarningDolphinFlip_45x42 which apparently is not defined. 

Commenting out the sections in the two files makes the program compile again and now I also get working 2FA codes.

The git diff is :

<pre><font color="#CC0000">-#if __has_include(&lt;assets_icons.h&gt;)</font>
<font color="#CC0000">-            dialog_message_set_icon(message, &amp;I_WarningDolphinFlip_45x42, 83, 22);</font>
<font color="#CC0000">-#else</font>
<font color="#4E9A06">+//#if __has_include(&lt;assets_icons.h&gt;)</font>
<font color="#4E9A06">+//            dialog_message_set_icon(message, &amp;I_WarningDolphinFlip_45x42, 83, 22);</font>
<font color="#4E9A06">+//#else</font>
             dialog_message_set_icon(message, &amp;I_DolphinCommon_56x48, 72, 17);
<font color="#CC0000">-#endif</font>
<font color="#4E9A06">+//#endif</font>
             dialog_message_show(plugin_state-&gt;dialogs_app, message);
             dialog_message_free(message);
         }
<b>diff --git a/totp/ui/scenes/standby/standby.c b/totp/ui/scenes/standby/standby.c</b>
<b>index 6ac4d65..eed7792 100644</b>
<b>--- a/totp/ui/scenes/standby/standby.c</b>
<b>+++ b/totp/ui/scenes/standby/standby.c</b>
<font color="#06989A">@@ -6,11 +6,11 @@</font>
 #include &quot;../../constants.h&quot;
 
 void totp_scene_standby_render(Canvas* const canvas) {
<font color="#CC0000">-#if __has_include(&lt;assets_icons.h&gt;)</font>
<font color="#CC0000">-    canvas_draw_icon(canvas, SCREEN_WIDTH - 50, SCREEN_HEIGHT - 44, &amp;I_WarningDolphinFlip_45x42);</font>
<font color="#CC0000">-#else</font>
<font color="#4E9A06">+//#if __has_include(&lt;assets_icons.h&gt;)</font>
<font color="#4E9A06">+//    canvas_draw_icon(canvas, SCREEN_WIDTH - 50, SCREEN_HEIGHT - 44, &amp;I_WarningDolphinFlip_45x42);</font>
<font color="#4E9A06">+//#else</font>
     canvas_draw_icon(canvas, SCREEN_WIDTH - 56, SCREEN_HEIGHT - 48, &amp;I_DolphinCommon_56x48);
<font color="#CC0000">-#endif</font>
<font color="#4E9A06">+//#endif</font></pre>

**Expected behavior**
totp.fap file that works on official firmware

**Device information (please complete the following information):**
 - Firmware: Official stable
 - Firmware version 0.99.1

akopachov, 2024-04-02:
Looks like [uFBT](https://github.com/flipperdevices/flipperzero-ufbt) builds it [just fine](https://github.com/akopachov/flipper-zero_authenticator/actions/runs/8521248935/job/23339016088). I guess you need to build firmware\SDK (or so) first and then build Authenticator.

dengesCU, 2024-04-03:
Thanks for the quick reply, I'll look into uFBT for building it in the future. The fap file from the nightly build also works on my flipper.  So problem solved :)

Just in case, this is how the build error is reproducible for me (with clean repos):
``` shell
git clone --recursive https://github.com/flipperdevices/flipperzero-firmware.git
git clone https://github.com/akopachov/flipper-zero_authenticator.git
cd flipperzero-firmware
git checkout release
ln -s ../../flipper-zero_authenticator/totp applications_user/totp
./fbt fap_totp
```

PS: Since I still don't know how to build correctly using fbt but it clearly works for you with uFBT, I'll leave the issue open, but please feel free to close it. Thanks again for the help and of course for the great app

akopachov, 2024-04-03:
Glad you like it!

Also, thank you for providing the steps to reproduce.

closed 2024-05-13 by akopachov