akopachov/flipper-zero_authenticator #230: [FEATURE] List app in the Flipper Application Catalog

ololobus, 2024-04-05:
Flipper Authenticator looks like a great app. It doesn't seem to violate any guidelines at first sight and has OSS license.

What do you think about listing it in the official Flipper Application Catalog? https://github.com/flipperdevices/flipper-application-catalog

[Guideline is here](https://github.com/flipperdevices/flipper-application-catalog/blob/main/documentation/Contributing.md)

Sorry if answer is already mentioned somewhere and I just missed it :)

akopachov, 2024-04-05:
Hey, that's a great question. [Here is] (https://github.com/akopachov/flipper-zero_authenticator/wiki/Why-it-is-not-listed-in-official-Flipper-Application-Catalog%3F) the answer why it is not there, and unlikely be there.
TL;DR
1. Flipper Catalog maintainers dictate how the code should be formatted in your repo
2. They are tried to dictate me what I can put to the description and what I should not.

ololobus, 2024-04-06:
I see, thanks for the quick response, although I've seen the FAQ, just didn't scroll till the end %)

Well, [Flipper-XFW](https://github.com/Flipper-XFW/Xtreme-Firmware) is unofficial firmware based on [Unleashed](https://github.com/DarkFlippers/unleashed-firmware), which removes regional radio restrictions. It's illegal in the most of the world, unfortunately. Although I understand how annoying it is, I also understand that they as a company may not want to take the risks :( For example, Apple has crazy history of removing apps from the store, which were 100% safe. Nobody fully knows their logic, but I guess Flipper authors want to be extra cautious.

Don't write this to say that it's normal, just when you run the private company you have to balance and you want it to continue existing, while such monopolists like Apple and Google can make you life a hell if they ban you from their platforms.

Anyway, wish you all the best and going to play with flipper authenticator soon :)

closed 2024-04-06 by ololobus

akopachov, 2024-04-07:
Well, of course it might be so. But let's consider two points here:

1. Why don't they announce it publicly in their TOS - "any links to anything mentioning custom firmware is not allowed". So that everybody knows that this is Flipper Devices official position. 
2. I put a link to the discord server, not to the custom firmware website. Neither I nor anybody alse force to install custom firmware in that discord. And if somebody were running official firmware asking Authenticator-related question there were zero problem answering them.
3. It still doesn't explain why I should keep the code in *my* repo formatted the way Flipper Devices wants. To let them review it? I see no problem them running any formatter they want right before review. OSS allows it to them. 

Overall, I have nothing against of TOS and following them. What I avoid dealing with is a TOS which are not written, not officially and publicly announced and instead existing in some ephemeral form.

ololobus, 2024-04-07:
Yeah, I agree, they should've been more open and clear about the reasons

Az-Neter, 2024-04-23:
This isn't about censorship, this is 100% about your attitude.
All your abstinence is doing is preventing the vast majority of flipper users from actually caring about anything you do or make.

It is not at all unreasonable to expect an app NOT to link directly to a server that provides illegal firmware functions. What's crazy is that you're calling this censorship when you were asked to remove the xfw link and you'd have been allowed back on the store.

The official sources repeatedly say that you may not discuss or openly encourage any firmware that has illegal features. The ONLY reason people download unleashed and xfw is for said illegal features.

What's your deal, mate?

akopachov, 2024-04-23:
Dude, read [this first](https://github.com/akopachov/flipper-zero_authenticator/issues/230#issuecomment-2041319096)

If this is not allowed - no problem, mention this explicitly and publicly so that everybody knew who initiated that. I don't want it to look like it is me who decided so. Is that so hard to do?
Any good explanation on why I should keep formatted my code in my repo the way flipper devices want?