akopachov/flipper-zero_authenticator #231: Unable to set a new tokken if a pin code is set[BUG]

GrafOrlok, 2024-04-20:
I'm unable to add a new tokken entry with the flipper interface if a pin code is set, when i try to save the new entry the screen ask me to input the pin code and the entry is not saved.

Steps to reproduce the behavior:
1. Set a pin code
2. select the option : Add new token
3. Click on Name
4. Input the name and save it
5. go to the Select option and input the tokken
6. When you click save the app ask for the pin code
7. Nothing is saved

**Device information (please complete the following information):**
 - Firmware: xtreme
 - Firmware version XFW-0053

akopachov, 2024-04-23:
Nice Catch! It happens when you are typing name\secret for longer than a minute. Going to fix it soon. Meanwhile, feel free to use CLI to add your tokens.

closed 2024-04-23 by akopachov

akopachov, 2024-04-23:
Should be fixed in [v5.14.0](https://github.com/akopachov/flipper-zero_authenticator/releases/latest)