akopachov/flipper-zero_authenticator #233: [BUG] Null pointer dereference on latest official firmware

RingoAl, 2024-05-29:
After running app and enter pin, Flipper crashes and show "NULL pointer dereference" error

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest official firmware (0.102.3)
2. Build app (I've done it wia ufbt: cd totp, ufbt update, ufbt build)
3. Run app
4. Enter pin
5. See crash and error

**Expected behavior**
App works fine

**Screenshots**
[image]

**Device information (please complete the following information):**
 - Firmware: Official stable
 - Firmware version 0.102.3

akopachov, 2024-05-29:
Have you installed and run Authenticator previously on your device?
Is there `/ext/apps_data/totp/totp.conf`?

RingoAl, 2024-05-29:
>Have you installed and run Authenticator previously on your device?
Yes
>Does /ext/apps_data/totp/totp.conf file exist?
Yes

akopachov, 2024-05-29:
Could you please share `Version` and `CryptoVersion` properties out of `/ext/apps_data/totp/totp.conf`?

RingoAl, 2024-05-29:
Filetype: Flipper TOTP plugin config file
Version: 12
CryptoVersion: 3

RingoAl, 2024-05-29:
Flipper freezes on black screen on latest dev (F309844E) after pin

akopachov, 2024-05-29:
So far can't reproduce it on my flipper running official stable (0.102.3).

If you delete `/ext/apps_data/totp/totp.conf` (**take backup of it before**) does app start normally?

RingoAl, 2024-05-29:
I've renamed totp.conf.
App starts normally, asks for pin setup and crashes with same error on "Add new token"

akopachov, 2024-05-29:
Did exactly the same steps - still working just fine on my side.
Let's try to use Authenticator from the [latest nightly build](https://github.com/akopachov/flipper-zero_authenticator/actions/runs/9286344794/artifacts/1548289716).

RingoAl, 2024-05-29:
This build works fine, thanks.

akopachov, 2024-05-29:
No problem. Probably you were building outdated code or so. In any case - glad it works fine for you now

closed 2024-05-29 by akopachov