akopachov/flipper-zero_authenticator #244: [BUG] Didn't works on Unleashed 078

the-davinci-code, 2024-10-25:
used to use your application, but at that time I had the Xtreme firmware installed, and everything worked without any problems. But now, when I switched to Unleashed, I encountered the issue that this application cannot be used on this firmware. The application icon does not load:
[image]
 When logging in, there is even a notification that the application may work unstable:
[image]
And most importantly, on the Unleashed firmware, it is impossible to add tokens using [Companion](https://github.com/akopachov/flipper-zero_authenticator-companion):
[image]

akopachov, 2024-10-25:
It looks like Authenticator app you have is out of date. It is sort of strange, given the fact that Unleashed has it prebuilt, so I assume you installed it manually somehow for some reason. Which is fine.

Try installing Authenticator from [Nightly builds here](https://github.com/akopachov/flipper-zero_authenticator/actions/workflows/nightly-build.yml), they are running every day and it allows to keep them up to date with the latest firmwares.

the-davinci-code, 2024-10-25:
Thank you, I will try your solution today. But this is very strange: do you mean that there is a version of Unleashed firmware where Authenticator is already installed? It seems I have installed all versions of Unleashed (default, c, e, r) and did not see your application in any of them.

akopachov, 2024-10-25:
that's correct. Authenticator is included even in Unleashed default. So it should be there in "Apps/Tools" category. If you have an issue with it, consider raising a question in Unleashed repo, probably the maintainer there would be able to explain what is wrong.

closed 2024-10-25 by akopachov