akopachov/flipper-zero_authenticator #245: [BUG]

iecxiv, 2024-11-30:
**Describe the bug**
I have noticed that if for example you link the badkb by bluetooth to windows and although it is not being used, if I try to link authenticator also by bluetooth, it does not match, in any of the profiles. it is necessary to unlink badkb and then if it allows. it's a bit cumbersome to have to unlink one to use the other. this does not happen on Android. I can link as many apps as I want, be it badkb, bttrigger, authenticator... i don't know what it may be due to, I'm just notifying in the hope that a solution will be given

**Expected behavior**
that allows authenticator links at the same time as the rest of the apps, as the others already do, just as it works on android. make it work on Windows
**Screenshots**

**Device information (please complete the following information):**
 - Firmware: [momentum]
 - Firmware version [08]

**Additional context**
only in Windows it is necessary to unlink badkb, for example, to be able to link authenticator.

i don't know if it's a bug or a feature request, I'm just notifying

akopachov, 2024-12-02:
Hey, I doubt I'd be able to do something on this, given that this is how Windows OS works. From Authenticator side, I did what I can (custom BT device name, custom BT MAC). If you have a chance, try on different PCs and some other OSes.

iecxiv, 2024-12-02:
your work is appreciated. on two Windows PCs, the same thing. on android fine, and I guess on Linux equally fine, although I haven't tried it.