akopachov/flipper-zero_authenticator #251: [FEATURE]Secure OTP storage

fennectech, 2025-03-21:
**Is your feature request related to a problem? Please describe.**
I’d like to see a secure method of storing OTP seeds. 

**Describe the solution you'd like**
I’d like a D-Pad pin that is used to encrypt and decrypt OTP seeds using real cryptography 

**Describe alternatives you've considered**
N/A

**Additional context**
Strong crypto should be used

akopachov, 2025-03-24:
Hey, thank you for suggestion.

I agree that having a 10 digit pinpad will make encryption stronger than current 4 digit one. However, I'd like to emphasize that even now, [real cryptography (to the best of F0 device has) is used](https://github.com/akopachov/flipper-zero_authenticator/wiki/Is-it-secure%3F) to encrypt all the OTP seeds. In any case, it is always necessary to consider all the limitations of F0 hardware, that will physically not allow the same level of security as you get in enterprise level solutions or software like KeePass.

I will implement 10 digit pinpad instead of the current one when time allows.

fennectech, 2025-03-28:
Extra key depth is all thats really needed then.  Im glad that PBFDK2 is used for key derivation Sent from my iPad