akopachov/flipper-zero_authenticator #260: [BUG] Maximum Token Size?

blu006, 2025-04-29:
**Describe the bug**
Tokens of an unknown maximum size can not be generated correctly.

**To Reproduce**
*** Companion App ***
Steps to reproduce the behavior:
1. Add a token using companion app
    * Token tested was 1370 bits or 274*base32 characters
2. Observe the token generating incorrect results

*** Configuration File ***
Steps to reproduce the behavior:
1. Edit the configuration file to use the full, unencrypted 274*base32 secret
2. Observe that the token generation is still incorrect

**Expected behavior**
The tokens generated should be correct.

An additional token utilizing a much shorter secret (16*base32 or 80 bits) did not have the same issue.  It generated correct OTP tokens.

**Screenshots**
(Censored screenshots can be posted...)

**Device information (please complete the following information):**
 - Firmware: unlshd-08e

**Additional context**
I am not sure why my friend decided to use a 1370 OTP secret for their application.

akopachov, 2025-04-29:
Thanks for reporting this issue. It is not that popular to use such a long private key, BUT this actually allowed me to fix the issue I wanted to fix long time ago.

closed 2025-04-29 by akopachov

akopachov, 2025-04-29:
The issue should be fixed in v5.18.0

blu006, 2025-04-30:
I tested this with the newest version, and it works.  Thanks for the quick update!